dragino/Lora #2: GPS PPS pin

Romeo-Golf, 2016-12-23:
In order to use timing information from GPS to NTP it is required to take the PPS into a GPIO pin on the Pi, am I reading the schematics correctly that this PPS pin is only wired to the external pin on the HAT?

dragino, 2016-12-23:
Yes, The PSS pin is now connected to the LED and a pin header. it is not connected to the HAT now.

dragino, 2017-01-05:
> Thanks, would it be possible to have this either directly connected to one of the gpio pins or with a jumper to enable this is used for the GPS Rx/Tx?
>
> It seems that a number of tutorials about using Raspberry Pi with GPS to feed NTP where the PPS pin has been connected to GPIO pin 18 (physical pin 12) so could it be worthwhile using that pin for maximum compatibility?

Hi, Romeo, 

 

Thanks for this info. we will also add this feature. Can you point me the link so we can have a test on it?

 

Best Regards,

Edwin