dragino/Lora #3: Is external GPS Antenna Actively powered?

Romeo-Golf, 2017-01-03:
Having read a user who had to add "a cable to feed 5V to the centre pin of the external GPS antenna connector to use an active antenna" I just wanted to check if the external GPS antenna socket is supposed to be powered?

https://www.tiny-dev.com/lora-p2p-communication-with-lopy-dragino-lora-gps-hat-for-raspberry-pi/

dragino, 2017-01-03:
The current version doesn’t support external active GPS antenna, you have to use none-active one (no need voltage). We will add that in future hardware version. 

 

https://www.tiny-dev.com/lora-p2p-communication-with-lopy-dragino-lora-gps-hat-for-raspberry-pi/

Romeo-Golf, 2017-01-05:
Thanks, it would be great if the future revisions of the board had the ability to provide +5v to support active antennas... Would it be possible to have an LED to indicate when power is being drawn?

Also is it worth having another issue thread to track the planned changes for the future revision? :)

barsznica, 2017-10-11:
Most new devices will be 3.3V based, so this option will need to be considered, too

dragino, 2017-10-11:
Version LoRa GPS HAT v1.4 already upgraded to supported external active antenna.