dragino/Lora #4: codeRate on LoraWAN sensor node

amaestrini, 2017-07-19:
Dear Dragino developer,
we are very interesting in your project and we are testing LoRaWAN Dragino.
We have a Lora App Server and Network server online and some gateways based on Raspberry Pi3 and a Concentrator (this one: https://wireless-solutions.de/products/radiomodules/ic880a.html) with the lora_pkt_fwd and lora-brige installed.
For the sensor node we use an Arduino UNO with a Dragino GPS/Shield (with LMIC library for LoRaWAN stack https://github.com/matthijskooijman/arduino-lmic) and this code to send data: https://github.com/dragino/Lora/blob/master/Lora%20Shield/Examples/lora_shield_ttn/lora_shield_ttn.ino

It works both in ABP and OTAA, we receive the data on the LoRa App Server sent by the Dragino node.

We are able to change spreading factor from 7 to 12 in ABP mode and not in OTA. May be with OTA is not possible to change spreading factor on the Dragino code given that is the network server that has pre-configured values? did you do some test with OTA and different spreading factor values?

Can I ask you moreover if you have an idea how to change the codeRate? (now is 4/5) Is it possible to change codeRate on the Arduino/Dragino code?

Thanks a lot
Andrea

dragino, 2017-07-20:
Hi, Andrea, 

The LoRaWAN is software stack which runs on LoRa Radio physical layer. the Code Rate is defined in LoRa Radio and iin the software you can modify the sx1276 register to support different code rate. 

So in OTAA, should be your gateway/server send a command to change the code rate. 

>>Can I ask you moreover if you have an idea how to change the codeRate? (now is 4/5) Is it possible to change codeRate on the Arduino/Dragino code?

you need to check the library. it is possible to change the coderade manually.

gregarican, 2017-07-20:
If you are using the RadioHead library (which I had some good success with), then there are a few pre-defined modem configs. With a few different coding rates. Here’s a link --> http://www.airspayce.com/mikem/arduino/RadioHead/classRH__RF95.html#ab9605810c11c025758ea91b2813666e3. You can directly control coding rates apart from these, by writing to the modem registers. Although documentation and implementation of this is rather tedious. I personally ran into issues when I strayed too far from defaults or pre-defined modem settings. Specifically, the sender and receiver wouldn’t seem to see each other…

Greg Kujawa
IT Manager

DIAMOND CELLAR Holdings, LLC
6280 Sawmill Road
Dublin, OH 43017
614-923-9515 phone
614-336-4555 fax
www.diamondcellar.com<http://www.diamondcellar.com>
Office Hours: M Tu W Th F

[cid:image002.png@01CF8981.56A564A0]<https://www.facebook.com/DiamondCellar>[cid:image003.png@01CF8981.56A564A0]<https://twitter.com/DiamondCellarCo>[cid:image004.png@01CF8981.56A564A0]<http://www.pinterest.com/diamondcellar/>[cid:image005.png@01CF8981.56A564A0]<http://www.yelp.com/biz/diamond-cellar-dublin>[cid:image006.png@01CF8981.56A564A0]<http://instagram.com/diamondcellar>[cid:image007.png@01CF8981.56A564A0]<https://plus.google.com/+DiamondCellarDublin/about>

CONFIDENTIALITY NOTICE
UNAUTHORIZED INTERCEPTION IS PROHIBITED BY FEDERAL LAW
[Electronic Communications Privacy Act of 1986, 18 U.S.C. 2701(a) and 2702(a)] This message is protected by Constitutional Rights and applicable legal privileges.  It is strictly confidential.

dragino, 2017-07-21:
In case use Radiohead, you can use our version https://github.com/dragino/RadioHead  , which is slight modified to support code rate modify; this is an example code to call them:
https://github.com/dragino/Arduino-Profile-Examples/blob/master/libraries/Dragino/examples/LoRa/LoRa_GPS_Track/track_GPS_Gateway/track_GPS_Gateway.ino

barsznica, 2017-10-11:
The RadioHead library is NOT LoRaWAN; it is "RawLoRa"!

closed 2017-10-11 by dragino