dragino/Lora #10: Unable to send to TTN raspberry pi shield

kenken64, 2019-03-02:
[image]

kenken64, 2019-03-04:
06:20:39.667 -> Stadata:
06:20:39.667 -> 133829: EV_TXCOMPLETE (includes waiting for RX windows)
06:20:39.667 -> Starting
06:20:39.667 -> Packet queued
06:20:39.667 -> Packet queued
06:20:39.667 -> LMIC.freq:911700000
06:20:39.667 -> 
06:20:39.667 -> 
06:20:39.667 -> Receive data:
06:20:41.802 -> 133792: EV_TXCOMPLETE (includes waiting for RX windows)
06:21:41.827 -> Packet queued
06:21:41.827 -> Packet queued
06:21:41.827 -> LMIC.freq:911900000
06:21:41.827 -> 
06:21:41.827 -> 
06:21:41.827 -> Receive data:
06:21:43.943 -> 4017441: EV_TXCOMPLETE (includes waiting for RX windows)

Hacklog24, 2019-07-09:
Hi,

I’m using a raspberry pi 3b and a Dragino gps/hat v1.4 as a sender. When I run the sample code https://github.com/ernstdevreede/lmic_pi with sudo, I get this error:

Failure
radio.c 664

this does not happen if I launch it normally. Could you help me?

Thank you!

asanz060, 2021-03-03:
Hi @Hacklog24 did you solve this issue? I am having the same error. Thank you!